wollewald/ICM20948_WE #1: Suggestion: Add getGyrOffsets() and getAccOffsets()

fralbo, 2021-08-07:
Hello,
It could be interesting to add a `getGyrOffsets()` and a `getAccOffsets()` methods because I think there is no necessity to compute new offsets every time we power it up.
So, we could save the data computed once and set it in the chip on subsequent power up.

wollewald, 2021-08-08:
@fralbo , first of all I have to say that I tried to work with the offset registers of the ICM-20948, but I couldn't make that work correctly. Finally I have implemented my own methods. 

The first is the more complex one where you determine the min / max raw values by turning the ICM20948 slowly. These are values you have to insert anyway into the setAccOffsets and the setGyrOffsets functions. Correction and offset factors are calculated from these min/max values. The correction values are for the slope (delta angle / delta raw) and the offsets are for the zero points. 

The autoOffsets() function is mainly for the measurement of angles up to 60°. It just ensures you start with 0° on the x and y-axis. autoOffsets optimizes the offsets just for this purpose. I would not recommend it for other purposes. It can even have a negative effect on x and y values at bigger angles, since the slope is not recalculated. 

So, internally are not only offsets used, but also correction values for the slope. I would need to implement another function to request these values (offset and correction) and another function to re-apply them. This is not much effort to add. But the two calibration methods are are anyway difficult to understand for many users. I fear that most people are lost when they find another function for calibration.

Happy to hear your thoughts.

fralbo, 2021-08-08:
Hello Wolfgang,

Thank for your reply, I'll think a bit more about it.
Did you try to experiment the filters?
I tried all of them about gyro values, I don't see any effect.
At the moment I use a filter library.
In fact I first used the Sparkfun's library with dmp but my project is to
read lean angle on a motorbike and during a turn, angles widely drift to 0.
I suspect that the auto gravity correction doesn't work at all on motorbike.
I tried without dmp but I encountered problems reading angle. I'm not sure
it came from the library or not.
Anyway, the angle values were very well filtered by the dmp but that's a
nightmare without it.
I didn't made much tests at the moment.
More 1 one week.

Regards,
Alain

wollewald, 2021-08-08:
Hi Alain,

I have started to look into the implementation of the DMP functions, but then I noticed that this would exceed my capacities. The only filters I have implemented are the noise filters. They work fine but doesn't help you.  

At least for my library I can say that the angle measurements which are solely based on the acceleration sensor will not work on a motorbike that is moving. If you are driving through a curve gravity and centrifugal forces will balance so that the x and y acceleration sensor will not detect a change. Or in other words: if you would detect an angle while driving through a curve you would fall. What you should see, is an increase of acceleration in z-direction. 

Not sure if I can help.

Regards, Wolfgang

fralbo, 2021-08-09:
Hi Wolfgang,

In fact I though to just use the gyro value.
I will have to find a way to fix the drift next, maybe by compass but that
a different thing.
I just want to get a correct lean angle measure for now.

Regards,

fralbo, 2021-08-09:
Ok, just understood looking at your code after your previous reply, why
setting gyro dplf doesn't change anything in angle waveform. Of course, if
you use acceleration sensor....

Regards,

wollewald, 2025-10-01:
I think I can close this now

closed 2025-10-01 by wollewald