wollewald/ICM20948_WE #4: Yaw Calculation

atocec, 2023-06-09:
Hi,
I am using **ICM20948_07_pitch_and_roll** and using that I am getting two rotational values/degrees change.
But When I do rotation on Surface then I do not see any change. How can we calculate Yaw in Example#7 and have rotation changes in all directions?

wollewald, 2023-06-09:
Hi, the issue with the yaw angle is that you can't calculate it from the (static) acceleration values when the sensor is positioned flat. The z-sensor will show 1g, x and y are 0. 

You could use the magnetometer values to detect the direction of the x- and y-axis. In theory, this is easy, in praxis it is less easy. The biggest hurdle is to write a reliable calibration routine. I tried a bit but did not get something which was working good enough. Therefore, my library only provides the raw magnetometer values. 

The other option is to constantly measure the gyroscope data. E.g. if you detect a rotational speed of let's say 120°/s for 0.5 seconds, you know that the yaw angle has changed 60°. But even if you measure at high frequency, you will always have certain error and this will sum up over time. 

The best option to calculate orientations and rotations is through quaternions (https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation)

You find this implemented e.g. in the great library of Kris Winer for the MPU9250: 

https://github.com/kriswiner/MPU9250/blob/master/quaternionFilters.ino

But sorry, this is stuff for which I would need much time to understand how it's working and for the implementation. Time that I do not have, and therefore I will not promise to work on this in near future.

wollewald, 2023-09-17:
Since there was no further comment I close this issue.

closed 2023-09-17 by wollewald