wollewald/ICM20948_WE #5: How to find Drift Angle?

fenilGhoghari, 2023-08-02:
Hi,

I am using ICM20948 with ICM20948_WE library and now i am able to get appropriate coordinators of Accelerometer, Gyroscope and Magnetometer one Level of calibration is okay and it's working.

Now i have tried roll,Pitch and Yaw example to find Drift Angle for my skateboard but i am not able to found.

So my issue with Drift Angle can you please give some idea of formula to drive Drift Angle using this all parameter.

I have try with the Vector Formation to drive Angle but i not able to drive this i juse able to drive only heading angle so,

Please suggest me some formula or tutorial for this.

Thanks

fenilGhoghari, 2023-08-03:
Hi, @wollewald can you please suggest me how to find drift angle?

wollewald, 2023-08-03:
Sorry, for the moment I can only provide the just the gyroscope values, magnetometer values, acceleration values and angles derived from this. I know that there a methods to combine the values to get the absolute orientation but for foreseeable time I won't have the bandwidth to get into this. 
I guess you are looking for something like this:
https://github.com/kriswiner/MPU9250/blob/master/quaternionFilters.ino
Or did I understand your request wrong?