wollewald/ICM20948_WE #9: Unable to achieve >1 kHz FIFO output (expecting 4.5 kHz accel & gyro)

being24, 2025-08-31:
Hello,

I am trying to use the ICM20948 FIFO to capture both accelerometer and gyroscope data at the maximum possible rate. According to the datasheet, both the accelerometer and gyroscope should support up to 4.5 kHz when DLPF is off, but I only see around 1 kHz effective output when reading from the FIFO.

## Hardware / Setup

- MCU: M5AtomS3 (ESP32-S3 based)
- Interface: SPI mode @ 5 MHz
- Library: ICM20948_WE

## My configuration

- Accelerometer: DLPF_OFF, SampleRateDivider = 0 (expecting 4.5 kHz)
- Gyroscope: DLPF_OFF, SampleRateDivider = 0 (expecting 4.5 kHz)
- FIFO enabled in continuous mode, storing both accel and gyro
- Magnetometer disabled
- No cycle mode (ICM20948_NO_CYCLE)

## Behavior

Even with these settings, the logged sampling rate (measured over time by counting FIFO data sets) stays close to ~1,000 Hz total, rather than the expected ~4.5 kHz for both accel and gyro.

## Code
```cs
#include <ICM20948_WE.h>
#include <M5Unified.h>
#include <SPI.h>

// M5AtomS3 SPI pins for ICM20948
const int CS_PIN = 5;     // CS -> G5
const int MOSI_PIN = 6;   // MOSI -> G6
const int MISO_PIN = 7;   // MISO -> G7
const int SCLK_PIN = 39;  // SCLK -> G8

// Hardware timer for FIFO reading
hw_timer_t* timer = NULL;
SemaphoreHandle_t timerSemaphore;
void IRAM_ATTR onTimer();

// ICM20948 instance
ICM20948_WE myIMU(&SPI, CS_PIN, true);
xyzFloat gValue, gyr;

void setup() {
  // Initialize M5
  auto cfg = M5.config();
  cfg.clear_display = true;
  M5.begin(cfg);
  M5.delay(500);

  M5.Display.setTextSize(1);
  M5.Display.setCursor(0, 0);

  // Initialize semaphore
  timerSemaphore = xSemaphoreCreateBinary();
  if (timerSemaphore == NULL) {
    M5.Log.println("Failed to create semaphore");
    while (1);
  }

  // Initialize SPI for ICM20948
  SPI.begin(SCLK_PIN, MISO_PIN, MOSI_PIN, -1);
  SPI.setDataMode(SPI_MODE0);
  SPI.setFrequency(5000000);
  SPI.setBitOrder(MSBFIRST);

  // Initialize ICM20948
  if (!myIMU.init()) {
    M5.Display.println("ICM20948 does not respond");
  } else {
    M5.Display.println("ICM20948 is connected");
  }

  // Configure ICM20948 for high-speed sampling
  myIMU.setSPIClockSpeed(5000000);

  // Accelerometer configuration
  myIMU.setAccRange(ICM20948_ACC_RANGE_2G);
  myIMU.setAccDLPF(ICM20948_DLPF_OFF);  // Bypass DLPF for maximum speed
  myIMU.setAccSampleRateDivider(0);     // No division for maximum rate

  // Gyroscope configuration
  myIMU.setGyrRange(ICM20948_GYRO_RANGE_250);
  myIMU.setGyrDLPF(ICM20948_DLPF_OFF);  // Bypass DLPF for maximum speed
  myIMU.setGyrSampleRateDivider(1);  // Divide by 2 to sync with accelerometer

  // Disable magnetometer
  if (myIMU.initMagnetometer()) {
    myIMU.setMagOpMode(AK09916_PWR_DOWN);
  }

  myIMU.enableCycle(ICM20948_NO_CYCLE);

  // FIFO configuration
  myIMU.enableFifo(true);
  myIMU.setFifoMode(ICM20948_CONTINUOUS);
  myIMU.startFifo(ICM20948_FIFO_ACC_GYR);

  M5.Display.println("ICM20948 configured");
  M5.Display.println("FIFO: Continuous mode");
  M5.Display.println("Rate: Acc max, Gyr/2");

  // Hardware timer setup (10ms interval)
  timer = timerBegin(1000000);
  timerAttachInterrupt(timer, &onTimer);
  timerAlarm(timer, 10000, true, 0);

  M5.Display.println("Timer: 10ms interval");

  // Find FIFO begin position
  myIMU.findFifoBegin();
}

// Timer interrupt function
void IRAM_ATTR onTimer() {
  BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  xSemaphoreGiveFromISR(timerSemaphore, &xHigherPriorityTaskWoken);
  if (xHigherPriorityTaskWoken == pdTRUE) {
    portYIELD_FROM_ISR();
  }
}

void readFifoData() {
  int16_t fifoCount = myIMU.getFifoCount();
  int16_t dataSets = myIMU.getNumberOfFifoDataSets();
  static uint32_t totalSets = 0;
  static uint32_t readCount = 0;

  if (dataSets > 0) {
    unsigned long timestamp = millis();
    totalSets += dataSets;
    readCount++;

    // Read accelerometer and gyroscope data from FIFO
    for (int i = 0; i < dataSets; i++) {
      myIMU.getGValuesFromFifo(&gValue);
      myIMU.getGyrValuesFromFifo(&gyr);
    }

    // Log output every 100 reads (approximately every 1 second)
    if (readCount % 100 == 0) {
      M5.Log.printf(
          "FIFO: %d sets, %d bytes | Total: %d sets, Read: %d times\n",
          dataSets, fifoCount, totalSets, readCount);

      // Calculate sampling rate
      static uint32_t lastTotalSets = 0;
      static uint32_t lastTimestamp = 0;

      if (lastTimestamp > 0) {
        uint32_t deltaSets = totalSets - lastTotalSets;
        uint32_t deltaTime = timestamp - lastTimestamp;

        if (deltaTime > 0) {
          float instantRate = (float)deltaSets * 1000.0 / deltaTime;
          M5.Log.printf("Sampling rate: %.1f Hz (last %d ms)\n", instantRate,
                        deltaTime);
        }
      }

      lastTotalSets = totalSets;
      lastTimestamp = timestamp;
    }
  }
}

void loop() {
  // Wait for timer semaphore (10ms interval)
  if (xSemaphoreTake(timerSemaphore, portMAX_DELAY) == pdTRUE) {
    readFifoData();
  }
}

```

```sh
FIFO: 12 sets, 144 bytes | Total: 16010 sets, Read: 1400 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 17153 sets, Read: 1500 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 18296 sets, Read: 1600 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 19439 sets, Read: 1700 times
Sampling rate: 1143.0 Hz (last 1000 ms)
```

## Question

Is there something wrong with my configuration, or is there a limitation in the library/hardware that prevents achieving the expected 4.5 kHz accel + 4.5 kHz gyro sampling rate from the FIFO?

Any advice or clarification would be greatly appreciated!

wollewald, 2025-09-01:
And now I have "translated" your code for an ESP32 development board:

````
#include <ICM20948_WE.h>
#include <SPI.h>

volatile bool timerEvent = false;
const int CS_PIN = 5;    
const int intPin = 22;

// Hardware timer for FIFO reading
hw_timer_t* timer = NULL;

void IRAM_ATTR onTimer() {
  timerEvent = true;
}

// ICM20948 instance
ICM20948_WE myIMU(&SPI, CS_PIN, true);
xyzFloat gValue, gyr;

void setup() {
    // Initialize ICM20948
  Serial.begin(115200);
  delay(2000);
  if (!myIMU.init()) {
    Serial.println("ICM20948 does not respond");
  } else {
    Serial.println("ICM20948 is connected");
  }

  // Configure ICM20948 for high-speed sampling
  myIMU.setSPIClockSpeed(5000000);

  // Accelerometer configuration
  myIMU.setAccRange(ICM20948_ACC_RANGE_2G);
  myIMU.setAccDLPF(ICM20948_DLPF_1);  // Bypass DLPF for maximum speed
  myIMU.setAccSampleRateDivider(2);     // No division for maximum rate

  // Gyroscope configuration
  myIMU.setGyrRange(ICM20948_GYRO_RANGE_250);
  myIMU.setGyrDLPF(ICM20948_DLPF_1);  // Bypass DLPF for maximum speed
  myIMU.setGyrSampleRateDivider(2);  // Divide by 2 to sync with accelerometer

  // Disable magnetometer
  if (myIMU.initMagnetometer()) {
    myIMU.setMagOpMode(AK09916_PWR_DOWN);
  }

  myIMU.enableCycle(ICM20948_NO_CYCLE);

  // FIFO configuration
  myIMU.enableFifo(true);
  myIMU.setFifoMode(ICM20948_CONTINUOUS);
  myIMU.startFifo(ICM20948_FIFO_ACC_GYR);

  Serial.println("ICM20948 configured");
  Serial.println("FIFO: Continuous mode");
  Serial.println("Rate: Acc max, Gyr/2");

  // Hardware timer setup (10ms interval)
  timer = timerBegin(1000000);
  timerAttachInterrupt(timer, &onTimer);
  timerAlarm(timer, 10000, true, 0);

  Serial.println("Timer: 10ms interval");

  // Find FIFO begin position
  myIMU.findFifoBegin();
}

void readFifoData() {
  int16_t fifoCount = myIMU.getFifoCount();
  int16_t dataSets = myIMU.getNumberOfFifoDataSets();
  static uint32_t totalSets = 0;
  static uint32_t readCount = 0;

  if (dataSets > 0) {
    unsigned long timestamp = millis();
    totalSets += dataSets;
    readCount++;

    // Read accelerometer and gyroscope data from FIFO
    for (int i = 0; i < dataSets; i++) {
      myIMU.getGValuesFromFifo(&gValue);
      myIMU.getGyrValuesFromFifo(&gyr);
    }

    // Log output every 100 reads (approximately every 1 second)
    if (readCount % 100 == 0) {
      Serial.printf(
          "FIFO: %lu sets, %lu bytes | Total: %lu sets, Read: %lu times\n",
          dataSets, fifoCount, totalSets, readCount);

      // Calculate sampling rate
      static uint32_t lastTotalSets = 0;
      static uint32_t lastTimestamp = 0;

      if (lastTimestamp > 0) {
        uint32_t deltaSets = totalSets - lastTotalSets;
        uint32_t deltaTime = timestamp - lastTimestamp;

        if (deltaTime > 0) {
          float instantRate = (float)deltaSets * 1000.0 / deltaTime;
          Serial.printf("Sampling rate: %.1f Hz (last %lu ms)\n", instantRate,
                        deltaTime);
        }
      }

      lastTotalSets = totalSets;
      lastTimestamp = timestamp;
    }
  }
}

void loop() {
  // Wait for timer semaphore (10ms interval)
  if (timerEvent) {
    readFifoData();
    timerEvent = false;
  }
}

````

Then I have played a bit with the DLPFs an sample rate dividers. Interestingly it has no effect, I always get 1134 Hz. Can you also try? Same on your side?

being24, 2025-09-01:
Thank you for your reply and verification.
I tried the code you provided (I only have an M5 Atom S3 on hand, so I modified some parts of the code).

The sketch is as follows:

```cs
#include <ICM20948_WE.h>
#include <SPI.h>

volatile bool timerEvent = false;
const int CS_PIN = 5;
const int intPin = 8;
const int MOSI_PIN = 6;   // MOSI -> G6
const int MISO_PIN = 7;   // MISO -> G7
const int SCLK_PIN = 39;  // SCLK -> G8

// Hardware timer for FIFO reading
hw_timer_t* timer = NULL;

void IRAM_ATTR onTimer() { timerEvent = true; }

// ICM20948 instance
ICM20948_WE myIMU(&SPI, CS_PIN, true);
xyzFloat gValue, gyr;

void setup() {
  // Initialize ICM20948
  Serial.begin(115200);
  delay(2000);

  // SPI初期化（ICM20948用）
  SPI.begin(SCLK_PIN, MISO_PIN, MOSI_PIN, -1);  // CSピンは-1（使用しない）

  // Set SPI mode 0 (CPOL=0, CPHA=0) for devices with CS pin
  SPI.setDataMode(SPI_MODE0);
  SPI.setFrequency(5000000);
  SPI.setBitOrder(MSBFIRST);

  if (!myIMU.init()) {
    Serial.println("ICM20948 does not respond");
  } else {
    Serial.println("ICM20948 is connected");
  }

  // Configure ICM20948 for high-speed sampling
  myIMU.setSPIClockSpeed(5000000);

  // Accelerometer configuration
  myIMU.setAccRange(ICM20948_ACC_RANGE_2G);
  myIMU.setAccDLPF(ICM20948_DLPF_1);  // Bypass DLPF for maximum speed
  myIMU.setAccSampleRateDivider(2);   // No division for maximum rate

  // Gyroscope configuration
  myIMU.setGyrRange(ICM20948_GYRO_RANGE_250);
  myIMU.setGyrDLPF(ICM20948_DLPF_1);  // Bypass DLPF for maximum speed
  myIMU.setGyrSampleRateDivider(2);   // Divide by 2 to sync with accelerometer

  // Disable magnetometer
  if (myIMU.initMagnetometer()) {
    myIMU.setMagOpMode(AK09916_PWR_DOWN);
  }

  myIMU.enableCycle(ICM20948_NO_CYCLE);

  // FIFO configuration
  myIMU.enableFifo(true);
  myIMU.setFifoMode(ICM20948_CONTINUOUS);
  myIMU.startFifo(ICM20948_FIFO_ACC_GYR);

  Serial.println("ICM20948 configured");
  Serial.println("FIFO: Continuous mode");
  Serial.println("Rate: Acc max, Gyr/2");

  // Hardware timer setup (10ms interval)
  timer = timerBegin(1000000);
  timerAttachInterrupt(timer, &onTimer);
  timerAlarm(timer, 10000, true, 0);

  Serial.println("Timer: 10ms interval");

  // Find FIFO begin position
  myIMU.findFifoBegin();
}

void readFifoData() {
  int16_t fifoCount = myIMU.getFifoCount();
  int16_t dataSets = myIMU.getNumberOfFifoDataSets();
  static uint32_t totalSets = 0;
  static uint32_t readCount = 0;

  if (dataSets > 0) {
    unsigned long timestamp = millis();
    totalSets += dataSets;
    readCount++;

    // Read accelerometer and gyroscope data from FIFO
    for (int i = 0; i < dataSets; i++) {
      myIMU.getGValuesFromFifo(&gValue);
      myIMU.getGyrValuesFromFifo(&gyr);
    }

    // Log output every 100 reads (approximately every 1 second)
    if (readCount % 100 == 0) {
      Serial.printf(
          "FIFO: %lu sets, %lu bytes | Total: %lu sets, Read: %lu times\n",
          dataSets, fifoCount, totalSets, readCount);

      // Calculate sampling rate
      static uint32_t lastTotalSets = 0;
      static uint32_t lastTimestamp = 0;

      if (lastTimestamp > 0) {
        uint32_t deltaSets = totalSets - lastTotalSets;
        uint32_t deltaTime = timestamp - lastTimestamp;

        if (deltaTime > 0) {
          float instantRate = (float)deltaSets * 1000.0 / deltaTime;
          Serial.printf("Sampling rate: %.1f Hz (last %lu ms)\n", instantRate,
                        deltaTime);
        }
      }

      lastTotalSets = totalSets;
      lastTimestamp = timestamp;
    }
  }
}

void loop() {
  // Wait for timer semaphore (10ms interval)
  if (timerEvent) {
    readFifoData();
    timerEvent = false;
  }
}
```
(I changed the pin number and modified the SPI definition.)

The results are as follows:

```sh
FIFO: 11 sets, 132 bytes | Total: 2286 sets, Read: 200 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 11 sets, 132 bytes | Total: 3429 sets, Read: 300 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 4572 sets, Read: 400 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 5715 sets, Read: 500 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 6858 sets, Read: 600 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 12 sets, 144 bytes | Total: 8001 sets, Read: 700 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 11 sets, 132 bytes | Total: 9143 sets, Read: 800 times
Sampling rate: 1142.0 Hz (last 1000 ms)
FIFO: 11 sets, 132 bytes | Total: 10286 sets, Read: 900 times
Sampling rate: 1143.0 Hz (last 1000 ms)
FIFO: 11 sets, 132 bytes | Total: 11429 sets, Read: 1000 times
Sampling rate: 1143.0 Hz (last 1000 ms)
```

The results are a bit different...

wollewald, 2025-09-01:
Very strange. At least I can say the following:

- Changing the sample rate for the FIFO basically works. This is proven by the results of the first sketch I have published above. 
- From ICM20948_13_FIFO_continuous.ino I know that changing the sample rate works also in the continuous mode.
- That problem has nothing to with my first thought that reading the values takes too much time. 
- I can't find any error in your code (and my "translation" for the ESP32)

Can't conclude anything that makes sense. Will think about it.

wollewald, 2025-09-01:
If you get stuck, it's always good to have a break and look at things with a fresh mind. I think I found the problem.  At least it works on my side.

Move this part of your code:
````
// Disable magnetometer
  if (myIMU.initMagnetometer()) {
    myIMU.setMagOpMode(AK09916_PWR_DOWN);
  }
````
before:  

````
// Accelerometer configuration
````

The initiation of the Magnetometer requires a reset of the ICM20948 which explains why the settings you made before are lost and have no effect.

being24, 2025-09-02:
Thank you so very much

It worked on my environment too!!!

wollewald, 2025-09-03:
Great! 

I like your program. Are you OK if I add it as an example to this lib?

closed 2025-09-04 by wollewald